Opentrons/opentrons #1406: PD: delete labware gracefully

IanLondon, 2018-05-09:
## desired behavior

When a labware that is used in a step is deleted, PD should show an error banner 'This action involves labware that has been deleted.'

## steps to reproduce

Set up a step that uses a labware, then go back into Deck Setup and delete that labware. PD white-screens.

howisthisnamenottakenyet, 2018-05-21:
@IanLondon I think steps that have deleted labware can do two things:
(1) Show the banner you suggested 
(2) Reset the labware and wells fields that used to contain info about that labware. 

Is this something you need a design for or can we move ahead?

IanLondon, 2018-05-21:
sounds good

IanLondon, 2018-05-22:
Closed by #1481 (also is duplicate of #964 )

IanLondon, 2018-05-22:
oops mixed this with delete step; still open

PR #1491 (2018-05-22): PD: delete labware gracefully
## overview

Closes #1406

## changelog

- Aspirate & dispense command creators return "labware does not exist" error object instead of raising an exception, when labware id does not exist in RobotState

## review requests

https://s3-us-west-2.amazonaws.com/opentrons-protocol-designer/pd_delete-labware-gracefully/index.html

- Create labware X
- Set up steps
- Go back to Deck Setup, delete labware X, you should see an error banner 'This step involves labware that has been deleted'
- Fix the step by selecting another labware Y, the error will go away

And for mult [... cut]
closed 2018-05-22 by IanLondon