Opentrons/opentrons #1496: Unable to Accurately Hit A1 Well of 96-Flat

umbhau, 2018-05-22:
Robot unable to accurately hit known labware

## Expected Behavior
- Robot should be able to hit center of A1 well in 96-flat during labware calibration

## Observed Behavior
- Multiple robots consistently move to the same wrong location relative to the A1 when during labware calibration

## Steps to Recreate
- Connect to any production robot using 3.0.0 app
- Perform deck calibration using CLI tool (tested including and not including mount-offset addition)
- Confirm robot has an accurate x-y labware definition for the 96-flat (see below) 
- Load a protocol in the run app including a 96-flat labware and a P300S pipette
- Perform tip probe and tip rack calibration
- Attempt to calibrate labware
- Note that pipette tip moves about 1.4 mm too far forward in the Y direction
- Repeat process on multiple robots with same result

## Labware File
Note this is the expected / correct labware definition for a 96-flat (x and y are confirmed; z is likely off)

shared_data > definitions > 96_flat.json

``` "wells": {
    "A1": {
      "depth": 10.5,
      "diameter": 6.4,
      "height": 10.5,
      "length": 6.4,
      "total-liquid-volume": 400,
      "width": 6.4,
      "x": 11.14, # bottom of 
      "y": 71.04, # bottom of plate to bottom of well
      "z": 0
    },```

umbhau, 2018-05-22:
Note this data was created with a single on the right mount

btmorr, 2018-05-24:
You included the data from a new labware definition, but I don't see in the directions that the new labware definitions feature flag is set. I'll work today on trying to diagnose

btmorr, 2018-05-24:
After commenting out tip probe, the robot goes to exactly the same point on repeated runs. My recommendation for the quickest solution is to make tip probe optional.

mcous, 2018-05-25:
Shifted this ticket over to the app based on talks with @btmorr and @umbhau

PR #1558 (2018-05-30): feat(app): Make tip probe completely optional
## overview

This PR makes tip probe completely optional for both protocol run and labware calibration

Closes #1496

## changelog

- feat(app): Make tip probe completely optional

## review requests

Virtual smoothie is fine. Making tip probe optional messes a little with the existing logic for when to display the various clear / populate deck modals, so should test that still works:

- [x] Can run a protocol without first doing tip probe
- [x] Can calibrate labware without first doing tip probe
- Ensure "clear deck" alert modal still pops up befo [... cut]
commit 8ddfb4c8 (2018-05-30): feat(app): Make tip probe completely optional (#1558)

Closes #1496
closed 2018-05-30 by mcous