Opentrons/opentrons #1536: Swap pipettes in PD

howisthisnamenottakenyet, 2018-05-24:
As a user I would like to be able to swap L and R pipettes in the file details page so I don't have to physically do it on the robot

## Acceptance criteria
-Add a button to file details that switches which pipette is on which side
-This updates the tip rack assignment text in dropdown accordingly

## Design
SVG called 'swap-horizontal'
[image]

pantslakz, 2018-07-02:
@howisthisnamenottakenyet I have a hunch the arrows might not be clear enough in this situation. Would suggest a button with text in place of icon here.

ie. "Swap pipette positioning"

howisthisnamenottakenyet, 2018-07-02:
@pantea how about something like this?
[image]

pantslakz, 2018-07-03:
@howisthisnamenottakenyet how would it look if both pipettes were populated?

PR #1883 (2018-07-18): feat(protocol-designer): enable user to swap pipette mounts
## overview

Closes #1536

Allows user to swap pipette mounts. If you set up a protocol with one or two pipettes, then swap the mounts, the protocol should function the same with no other changes required. Ie, you can just click "SWAP", and swapping shouldn't ever require the user to make any other changes to the steps etc.

## changelog

- Add `swap-horizontal` icon
- Add `contentClassNameOverride` prop to `Card` component (this is so I can opt out of `display: flex`)
- Use `Card`s to organize FilePage contents to match design
- Add  [... cut]
commit d5e40cdc (2018-07-18): feat(protocol-designer): enable user to swap pipette mounts (#1883)

Closes #1536
closed 2018-07-18 by IanLondon