Opentrons/opentrons #1561: PD Bugs: Unable to view File page when editing ingredients

pantslakz, 2018-05-29:
As a user, I'd like to view the File page even when I'm editing the ingredients

## Expected Behavior
- View File page when you select 'File' tab in primary nav panel

## Observed Behavior
- Deck setup persists after user selects 'File' tab.

## Steps to recreate 
- add ingredient to labware
- select 'File' tab

IanLondon, 2018-05-31:
@pantslakz @howisthisnamenottakenyet 

When a user clicks the File page, then clicks back to the Design page, should whatever modals they had open stay open, or should they be closed when you click File page nav icon?

We should make a high-level flow chart of all the "modes" and modals in PD, and what side-effects happen when the user navigates between them

IanLondon, 2018-10-31:
Covered in #2198

closed 2018-10-31 by IanLondon