Opentrons/opentrons #1574: App Update: Linux support

mcous, 2018-05-31:
## overview

Our current Linux build (`.deb` target) does not support:

1. Retrieving the latest version number of the app
2. Updating itself

(2) is a limitation of the `deb` target of `electron-builder` (traditionally, `deb` application updates are managed through the system's package manager. (1) however, is totally fixable.

## acceptance criteria

- [ ] App on Linux is able to retrieve the latest version number
- [ ] Update modal informs user that auto-update is unavailable
- [ ] Update modal should provide a link to the updated app file
    - This should also be the behavior of Mac and Windows if autoupdate fails

## implementation details

- May involve also publishing an `AppImage` target
    - Enabling `AppImage` in `electron-builder`will result in the publishing of `latest-linux.yml`
        - Update metadata file
        - Includes version number and update file
    - We don't have to publicize the AppImage build, but the `deb` app can still read the update metadata

## design

@Opentrons/ux engineering can wing it on the modal (by re-using the existing one) but could use your help with copy. Feel free to hit me up on Slack or something for more background into this issue and proposed solution

btmorr, 2018-09-13:
@mcous is this ticket still relevant? If so, should we prioritize it, or is it ok for it to live in the icebox for a while?

mcous, 2018-09-18:
Superseded by #2303

closed 2018-09-18 by mcous