Opentrons/opentrons #1587: link to documentation in PD nav

howisthisnamenottakenyet, 2018-06-04:
As a user I'd like to know where to access support documentation for PD

## Acceptance criteria
- In the nav under the design tab there should be an icon that goes to wherever we decide to document alpha

## Design
[image]

umbhau, 2018-06-08:
@howisthisnamenottakenyet do we want to follow Run App pattern and put the app info/help tab at the bottom of the primary nav?

howisthisnamenottakenyet, 2018-06-08:
@umbhau Yup. Was talking it over w Pantea. Going to update the wires to stuff it in a More drawer too.

IanLondon, 2018-07-26:
Blocked: we're not sure if Intercom messenger should be in PD

IanLondon, 2019-01-11:
@umbhau are we good to close this ticket?

mcous, 2019-06-28:
Closing in favor of a new ticket if this ever gets revisited

closed 2019-06-28 by mcous