Opentrons/opentrons #1603: Warning: aspirate volume is less than recommended for your pipette

howisthisnamenottakenyet, 2018-06-04:
As a user I'd like to know if I chose an aspirate volume that's below a pipette's recommended range

## Acceptance criteria
- In this scenario generate a modal

## Design
- Title: Aspirate volume is less than recommended for your pipette
- Body: The recommended range for a {pipette type} is {range}

umbhau, 2018-06-08:
@howisthisnamenottakenyet 

Why a modal for this vs. the alert bar we use in #1089 and #1090?

Also -- assuming we'll catch this as a field-level validation in most cases, unless it happens as a downstream effect of another change?

howisthisnamenottakenyet, 2018-06-11:
@umbhau Was a typo! Should totally be an alert.

umbhau, 2018-12-06:
fyi, transfer plan: 
https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/legacy_api/instruments/pipette.py#L1476

commit eee9cf94 (2018-12-11): feat(protocol-designer): avoid transfer below pipette min volume

* break up transfers so no chunk goes under min volume
* change steplist volumes to use 2
significant digits
* form-level warning when specified volume is below pipette max

Closes #1603
commit 5f9eba85 (2019-01-08): feat(protocol-designer): avoid transfer below pipette min volume

* break up transfers so no chunk goes under min volume
* change steplist volumes to use 2
significant digits
* form-level warning when specified volume is below pipette max

Closes #1603
PR #2804 (2019-01-08): feat(protocol-designer): avoid aspirate/dispense below pipette min volume
## overview

Closes #1603

## changelog

* break up transfers so no chunk goes under min volume
* change steplist volumes to use 2 significant digits
* form-level warning when specified volume is below pipette min

## open question

Should there be timeline-level warnings when you put in a volume that's below the pipette minimum? Eg if you try to transfer/distribute/consolidate/mix 10uL with a P300 etc?

## review requests

NOTE: this does not fix the existing bug where newly-made forms don't show form-level errors relat [... cut]
commit 2430e090 (2019-01-08): feat(protocol-designer): avoid aspirate/dispense below pipette min volume (#2804)

* add form-level error for volume below pipette minimum
* break up transfers so no chunk goes under min volume
* change steplist volumes to use 2 significant digits
* form-level warning when specified volume is below pipette max

Closes #1603
closed 2019-01-08 by IanLondon