Opentrons/opentrons #1610: Errors: File upload

howisthisnamenottakenyet, 2018-06-05:
- [ ]  ### This isn't a JSON or .py file

- [ ]  ### This isn't a .py file

- [ ]  ### This is a JSON file but there are errors so we can't open it 

- [ ]  ### This isn't a valid JSON file but not made by PD. Might be able to run fine on robot.

umbhau, 2018-07-02:
Let's collapse error types:
1) Incorrect file type (and call out that we only support JSON + don't support the old .py files)
2) Invalid protocol file (correct file type but invalid protocol. including invalid pipetting steps, metadata, etc. can call out that files not created in protocol designer may not load)

howisthisnamenottakenyet, 2018-07-02:
@umbhau Do you think this copy covers our cases?

umbhau, 2018-07-02:
@howisthisnamenottakenyet -- one minor thing: 
"Only JSON files created in the Protocol Designer can be imported" and "At this time we do not support JSON files created outside of the Protocol Designer" aren't strictly true -- you _could_ create a JSON protocol outside the designer, as long as you followed the right formatting guidelines.

Maybe something like the below?

"Unsupported File Type
Please note the Protocol Designer only supports Opentrons 'JSON' protocol files (look for the .json extension at the end of the filename). "

"Invalid File Format
This protocol may have missing information or formatting errors. For best results, we recommend always using the Opentrons Protocol Designer to edit and save protocol files."

PR #1829 (2018-07-10): feat(protocol-designer): show file upload errors in modal
## overview

Closes #1610

## changelog

- `load-file` atom has its first and currently only reducer, which holds file upload errors and is used by `FileUploadErrorModal`
- `loadFile` action converted from a simple action to a thunk, which handles file upload & parsing. It takes the `onChange` event from an `<input type='file' />` component, and dispatches either FILE_ERRORS or LOAD_FILE action.
- show file upload errors in new modal

## review requests

- Uploading a file that doesn't end in `*.json` will give you the "incorrect file t [... cut]
commit 5ffed813 (2018-07-10): feat(protocol-designer): show file upload errors in modal (#1829)

Closes #1610
closed 2018-07-10 by IanLondon