Opentrons/opentrons #1613: Show pipette moving + spinner overlay on deck calibration exit

pantslakz, 2018-06-05:
## overview

When exiting deck calibration: it doesn’t have the same pipette moving + spinner overlay as when you start calibration. Because the exit modal just persisted after clicking exit I wasn’t sure if that was going right. 

## steps to reproduce

- [ ] Start deck calibration 
- [ ] Exit deck calibration

mcous, 2018-06-06:
Reclassified as a feature because:

- App was implemented as spec'd by screens
- Robot responds immediately with a success message to a restart request, meaning tracking robot restart state in the app is too difficult

@pantslakz could you add one more screen to the Deck Calibration Zeplin project for spinner copy during restart?

@btmorr I think we need to discuss the `/restart` endpoint a little bit to figure out the best path for both displaying the fact that the robot is about to go / currently going down, and how to track when the robot comes back up properly.

mcous, 2018-06-25:
I think I misunderstood this ticket; this seems to be about early exit from deck calibration. Reclassified ticket back to an app bug accordingly.

commit df41d74e (2018-06-25): fix(app): Show spinner during home on deck calibration exit

Closes #1613
PR #1760 (2018-06-26): fix(app): Show spinner during home on deck calibration exit
## overview

This PR reorganizes the action chain on "Exit" click during deck calibration so that the homing spinner from #1726 gets shows during DC exit.

Closes #1613

## changelog

- fix(app): Show spinner during home on deck calibration exit 

## review requests

Required some tweaks in the bespoke action chaining util to handle the fact that the (now deprecated) react-router-redux middleware swallows route change actions rather than returning them, which goes against redux conventions and results in `dispatch(push(path))` returni [... cut]
commit b6999a86 (2018-06-26): fix(app): Show spinner during home on deck calibration exit (#1760)

Closes #1613
closed 2018-06-26 by mcous