Opentrons/opentrons #1623: Add date metadata to file details

howisthisnamenottakenyet, 2018-06-05:
As a user I'd like to know a bit about file history and when I last saved my work

## Acceptance criteria
- Add new data that shows up in file details for an imported file: "Date created"
- Add new data that shows up in file details for a file that was imported: "Last exported"
--- Display time + timezone as well. Might be helpful if you're repeatedly saving work. 

## Design
[image]

umbhau, 2018-06-08:
@howisthisnamenottakenyet @IanLondon 

Should this be 'Last Exported' or 'Last Modified'? I could see exports being a little weird to track, and a little confusing if you have multiple people passing a protocol file around.

umbhau, 2018-06-08:
@IanLondon defer to you on JSON date format, though probably something human readable?

howisthisnamenottakenyet, 2018-06-08:
@umbhau Last exported. One of its purposes at this time is to help you track when you last saved your work.

umbhau, 2018-06-10:
@IanLondon epoch it is (so yes to 1, 2, and 3)

PR #2754 (2018-12-05): feat(protocol-designer): show file created and modified date
## overview

Closes #1623 

## changelog

* Add moment.js to PD (just for datetime string formatting :cry:)
* Add "Date Created" and "Last Exported" fields
* Replace "unsaved file metadata fields" stuff in Redux with a Formik form

## review requests

* File Information card only lets you click "Update" button when changes have been made to the form
* Exporting a file will update the "Last exported" datetime immediately

* Is it OK that changing the fields then exporting the file without clicking "Update" will reset the form? Seems l [... cut]
commit 7fe3f0fc (2018-12-05): feat(protocol-designer): show file created and modified date (#2754)

* make metadata be a Formik form; remove unsaved metadata fields from redux

Closes #1623
closed 2018-12-05 by IanLondon