Opentrons/opentrons #1632: Run App: User Settings

umbhau, 2018-06-06:
As a user, I would like access to settings for my Robot and Run App.

## Acceptance Criteria
- User can view 'Advanced Settings' section on Robot Settings page
- User can view 'Advanced Settings' section on App Settings page
- User can toggle robot-level feature flags in the Robot 'Advanced Settings' section
- User can toggle app-level feature flags in the App 'Advanced Settings' section

## Design
- TBD @pantslakz

mcous, 2018-06-06:
@pantslakz for the moment, the only one we need for **app** is:

- Show dev tools (requires restart)

pantslakz, 2018-06-07:
@mcous when user flicks this on - what do they see in the app? Do we require additional UI within the app to represent these dev tools?

mcous, 2018-06-07:
@pantslakz, no, this just switches on the normal Chrome/Electron devtools. This is the same option that is automatically enabled when you run the application in development mode from source, so no additional UI is required

pantslakz, 2018-06-07:
from ben:
https://github.com/Opentrons/opentrons/blob/edge/api/opentrons/config/feature_flags.py

btmorr, 2018-06-08:
This looks fantastic. For the record, @pantslakz and I talked f2f, and the server is going to report a list of settings to expose through this panel in roughly the form:
```
{
  "settings": [
    {
      "title": "Calibrate to bottom of well",
      "description": "Perform labware calibration by moving the tip to the bottom-center of the well rather than the top-center (which is the default if this option is unset)",
      "value": true
    },
    {
      "title": "Other setting",
      "description": "Long description of next setting",
      "value": false
    }
  ]
}
```
@mcous do you have any thoughts/requests?

mcous, 2018-06-08:
That response schema looks perfect

btmorr, 2018-06-08:
Ticket for the backend work: #1656

mcous, 2018-06-11:
JK still need to do robot settings

umbhau, 2018-06-19:
@btmorr linking a ticket for blocker.

commit 6676903f (2018-06-11): feat(app): Toggle devtools feature flag in app settings (#1678)

Closes #1632
PR #1795 (2018-07-02): feat(app): Add robot settings toggles to Advanced Settings card
## overview

This PR is the front-end companion to #1786. It wires up `/settings` to the app.

Closes #1632

[image]

## changelog

- feat(app): Add robot settings toggles to Advanced Settings card 

## review requests

Sorry this PR is so large; it's mostly API client boilerplate and tests. I started work on a boilerplate reduction in this PR by creating generic API action creators (see `http-api-client/actions.js`). There's a path forward where all the reducers in each client module (e.g. `calibration`) are removed in favor of a [... cut]
commit 73f75289 (2018-07-02): feat(app): Add robot settings toggles to Advanced Settings card (#1795)

Closes #1632
closed 2018-07-02 by mcous