Opentrons/opentrons #1641: Deck Calibration: App should display robot errors

mcous, 2018-06-07:
## overview

As a user, I would like to know if a step during deck calibration fails so I can start the process over and try again.

## acceptance criteria

If the robot responds with an error to any request during deck calibration, the app should:

- [ ] Display the error message to the user (in case they need to contact support about it)
- [ ] Drop the user back to the beginning of the process

### current behavior

Right now, if the robot responds with an error, the app captures and stores the error internally, but then naively moves the user onto the next step in the calibration process, which is incorrect.

## implementation details

Many of the "proceed to next step" actions in Deck Calibration are thunked promise chains. Because of the way they are implemented, we don't currently have a mechanism to execute a chain of thunk actions but bail out if one of those actions results in a HTTP failure. This is easier to show than to tell, so hit @mcous up in person if you're curious. Technical side of this ticket will involve building this functionality or pulling it in from an external library.

@pantslakz we could use a screen / copy for displaying a generic error message along with a button to exit deck calibration (and possibly a button to restart deck calibration, but that could also be accomplished by dropping them back to robot settings where the user can click on the Deck Calibration button again.

Related to #1613

mcous, 2018-06-07:
@pantslakz would something like this work?

[image]

pantslakz, 2018-06-08:
Hey Mike, error modal looks correct - but how about something like:

(HEADER) Unexpected Error

(ERROR TYPE) Failed to fetch

(BODY) An unexpected error has cleared your deck calibration progress, please try again.

If you keep getting this message try restarting your robot. If this does not resolve the issue, please contact Opentrons Support.

mcous, 2018-06-08:
Prop details for implementation

> (HEADER) Unexpected Error

Hardcode for now, could eventually be a prop that defaults to "Unexpected Error"

> (ERROR TYPE) Failed to fetch

Existing prop `error.message`

> (BODY) An unexpected error has cleared your deck calibration progress, please try again.

Can hardcode for now. Will need to be a prop if we pull `ErrorModal` up out of `CalibrateDeck`

> If you keep getting this message try restarting your robot. If this does not resolve the issue, please contact Opentrons Support.

Hardcode. Will probably always be present in a generic error modal.

> (BUTTON) "Try Again"

@pantslakz "Try Again" seems to me to imply that clicking the button will drop you back to the beginning of the Deck Calibration process. However, current button behavior closes the modal and drops you back to Robot Settings. Do you have any thoughts here

pantslakz, 2018-06-11:
@mcous what about adding two buttons:
"Try again"  and  "Close"

mcous, 2018-06-12:
> This way they can avoid having to go through the 'place tip' screen and try saving first position again.

After discussing the technical side of things with @Kadee80, it was much easier with just "Close" for now. We can look into switching it to "Try Again" once we have "Close" in place. That being said, skipping the "place tip" screen isn't really possible, because the app confirming to the robot that there is a tip on is required for the robot to accurately track tip state. If the API errors out, we don't know if the API still has the correct tip-tracking state, and we don't have any mechanism for the app to ask the API about tip state at the moment.

pantslakz, 2018-06-12:
Ah ok - I assumed if the user gets an error mid process - they would still have the tip on, and could then proceed back to first calibration point.

I assume we have this same issue with tip probe (if user fails halfway - tip remains on when they restart).

I'm cool with coming back to this again later.

commit 9db70e9c (2018-06-11): feat(app): Capture and display errors during deck calibration

Closes #1641
PR #1680 (2018-06-12): feat(app): Capture and display errors during deck calibration
## overview

This PR adds some chaining logic to the API calls in deck calibration to ensure that if a given API call errors, deck calibration is stopped.

Closes #1641

## changelog

- feat(app): Capture and display errors during deck calibration 

## review requests

First off, don't be scared by the diff count. It's mostly a unit test file. Second off:

1. Start deck calibration with a robot
    - This doesn't work well with Virtual Smoothie without `driver_3_0.py` modifications
2. Induce some sort of failure (e.g. turn the robot [... cut]
commit 4f75ed7c (2018-06-12): feat(app): Capture and display errors during deck calibration (#1680)

Closes #1641
closed 2018-06-12 by mcous