Opentrons/opentrons #1656: User Settings: Expose user settings in server

btmorr, 2018-06-08:
## overview

Run app should be able to query the server for a list of settings to include in the "Advanced Settings" page.

## behavior

Schema:
```
{
  "settings": [
    {
      "title": "Calibrate to bottom of well",
      "description": "Perform labware calibration by moving the tip to the bottom-center of the well rather than the top-center (which is the default if this option is unset)",
      "value": true
    },
    {
      "title": "Other setting",
      "description": "Long description of next setting",
      "value": false
    }
  ]
}
```

There should be a GET endpoint that returns a title, long description, and current value for each setting in "feature_flags", and any other settings that we wish to expose through the advance settings page.

mcous, 2018-06-28:
Here's my pass at a spec. @btmorr and @Laura-Danielle let me know what you think

## fetch settings

### request

`GET /settings`

### response

```json
{
  "settings": [
    {
      "id": "short-identifier-1",
      "title": "User-facing title",
      "description": "User-facing description",
      "type": "boolean",
      "value": true
    },
    {
      "id": "short-identifier-2",
      "title": "User-facing title",
      "description": "User-facing description",
      "type": "number",
      "value": 42
    }
  ]
}
```

## set settings

### request

`POST /settings`

#### body

```json
{
  "id": "short-identifier-1",
  "value": false
}
```

### response

(Same response as `GET /settings`)

```json
{
  "settings": [
    {
      "id": "short-identifier-1",
      "title": "User-facing title",
      "description": "User-facing description",
      "type": "boolean",
      "value": false
    },
    {
      "id": "short-identifier-2",
      "title": "User-facing title",
      "description": "User-facing description",
      "type": "number",
      "value": 42
    }
  ]
}
```

btmorr, 2018-06-29:
Omitting "type" field for now--currently only supporting booleans

commit 7f094e10 (2018-06-29): feat(api): Add advanced settings endpoints to api server

Fixes #1656
commit 865f866c (2018-07-02): feat(api): Add advanced settings endpoints to api server

Fixes #1656
commit b2f8614e (2018-07-02): feat(api): Add advanced settings endpoints to api server

Fixes #1656
PR #1786 (2018-07-02): feat(api): Add advanced settings endpoints to api server
## overview

Add advanced settings endpoints to api server. Also added an lru_cache decorator to the "splitLabwareDefinitions" feature flag to reduce unnecessary disk IO (this is preferred anyway, as changing from old to new labware definitions without a reboot causes exceptions). 

Fixes #1656

## review requests

Test on robot that previously had a feature flag set to True, ensure that it remains True in new endpoint
closed 2018-07-02 by btmorr