Opentrons/opentrons #1660: Change edit labware nickname icon to pencil

howisthisnamenottakenyet, 2018-06-08:
As a user I might miss that there's an option for that in the sidebar, because it's the same icon in the same place as deck setup. Hopefully changing the icon will draw the eye better. 

## Acceptance criteria
-Switch icon from labware to edit pen

## Design
-Swap out icon here
[image]

umbhau, 2018-06-10:
@howisthisnamenottakenyet In the other sidebar cards, the icon represents what the card is (a transfer has an error, end has a checkmark). Any way we can keep this consistent?

e.g. an edit icon where other steps have a carat to expant? or something smarter than that idea.

howisthisnamenottakenyet, 2018-06-11:
@umbhau I don't know what you mean. Swing by my desk when you have time.

pantslakz, 2018-06-14:
another idea could be to treat the 'deck setup' card in sub nav as an expandable one which lists added labware in an accordion. labware is listed as labware type (not nickname)

if you click on any labware from the accordion or from the deck map - you arrive at the dark overlay with the appropriate labware diagram, and the nickname field above it. When you select a well, the nickname field turns into the ingredient fields.

as ingredients are added, they are added to the accordion in sub nav. 
you can change the nickname of the labware by clicking into it, and editing it in a field above the labware diagram.

howisthisnamenottakenyet, 2018-06-14:
@umbhau About swapping out the carat for an action button- I don't want to make a new way for the list item to behave for a one off without thinking about how it'll affect the component more broadly. 

I'm super open to changing components but I'd prefer it to be considered in context. If the list component isn't meeting our needs anymore let's definitely do a sit down next week and figure out how it should be changed.

PR #1842 (2018-07-10): feat(protocol-designer): change edit labware nickname icon to pen
## overview

Closes #1660

## changelog

## review requests

:pen:
commit 512f62c0 (2018-07-10): feat(protocol-designer): change edit labware nickname icon to pen (#1842)

Closes #1660
closed 2018-07-10 by IanLondon