Opentrons/opentrons #1662: Stop users from giving the same nickname to different labware

howisthisnamenottakenyet, 2018-06-08:
As a user building steps, I'd like to know which labware I'm selecting. Because of this labware should not be allowed to share nicknames. 

## Acceptance Critieria
@IanLondon What's the easiest way of handling this? An error? Or can we just start adding numbers like we do with labware type names. 
i.e Nickname (1), Nickname (2)

IanLondon, 2018-06-08:
@howisthisnamenottakenyet Both are close enough in terms of work to implement that I think we should do whatever is a better experience.

Seems like the "disambiguation number" `Nickname (2)` is the most straightforward to me, and it follows the pattern that the non-nicknamed labwares use

IanLondon, 2018-06-08:
This problem of sharing names can also happen when creating ingredients

howisthisnamenottakenyet, 2018-06-11:
@IanLondon I'm a fan of the numbering version too. 

For ingredients if you give something the same name isn't it just treated as the same ingredient? Considering wells with the same ingredient can have different volumes and different descriptions.

IanLondon, 2018-10-04:
@howisthisnamenottakenyet this is closed, right?

mcous, 2019-06-25:
Closing this because I assume it was closed by #3600

closed 2019-06-25 by mcous