Opentrons/opentrons #1676: Disposal Volume blowout destination

howisthisnamenottakenyet, 2018-06-11:
As a user I would like to be able to specify where my disposal volume ends up

## Background
- For disposal volume sometimes people want to save the liquid by doing the blow out not in trash. 

## Acceptance Criteria (WIP)
- When disposal volume in advanced settings is checked two things should appear:
(1) A volume field
(2) A blowout destination dropdown
- Options in the dropdown are
(1) Source well
(2) Trashes

## Design (WIP)

IanLondon, 2018-07-12:
Should the `Dispense: Blow out` in Distributes then go away? It would be implied that in Distributes, blow out must happen when you have a `Disposal Volume` (could also apply to any other future steps that have `Disposal Volume`)

howisthisnamenottakenyet, 2018-07-13:
@IanLondon I guess so! I'm trying to think of why someone might use both or want to use just blow out but no disposal volume....but I can't.

PR #2295 (2018-09-19): feat(protocol-designer): allow user to specify disposal volume dest

## overview

Instead of defaulting disposal volume to blowout in the fixed trash, we now give the user the option
to blowout their disposal volume from a distribute step into valid disposal labware as well as the
step's source well.

Closes #1676

## changelog

- Distribute step forms no longer have a dispense blowout option, rather their disposal volume has a blowout field to specify where to blowout the extra disposal volume
- You can now choose from "fixed trash", "source well", and any other acceptable trash box on the de [... cut]
commit 92ba8456 (2018-09-19): feat(protocol-designer): allow user to specify disposal volume dest (#2295)

Instead of defaulting disposal volume to blowout in the fixed trash, we now give the user the option
to blowout their disposal volume from a distribute step into valid disposal labware as well as the
step's source well.

Closes #1676
closed 2018-09-19 by b-cooper