Opentrons/opentrons #1692: Y axis slow during calibration/tip probe

jshmu, 2018-06-13:
## overview

## current behavior

## steps to reproduce

[QC_protocol_300m_300s.txt](https://github.com/Opentrons/opentrons/files/2099632/QC_protocol_300m_300s.txt)
[384_300s.txt](https://github.com/Opentrons/opentrons/files/2099633/384_300s.txt)

## expected behavior

mcous, 2018-06-13:
I was unable to reproduce this issue with the steps given above on a robot running 3.1.2. @jshmu did this issue happen more than once?

@btmorr or @andySigler do you know of anything transient off the top of your head(s) that could cause this behavior?

mcous, 2018-09-24:
Closing this issue since we couldn't repro and it's been a while. Will re-open if it is observed again

closed 2018-09-24 by mcous