Opentrons/opentrons #1694: Protocol pauses until user hits 'Resumes'

pantslakz, 2018-06-14:
As a user, I would like to include a pause command in my protocol, which continues only when I hit "resume" in Run App.

## Acceptance Criteria
- User able to pause a run from within protocol
- sees pause in RUN LOG with some kind of custom message (ie. "Add re-agent to well 4 and then hit resume") either in RUN LOG or Pause Banner
- User hits "resume" button in Run App

mcous, 2018-06-14:
@pantslakz to clarify, "User able to pause a run from within protocol" and "User hits 'resume' button in Run App" acceptance criteria already work with existing protocols. So that means this ticket is specifically for showing a more detailed message in the run log or in a banner, correct?

pantslakz, 2018-06-15:
@mcous yes!  @IanLondon can provide more detail relating to Protocol Designer.

mcous, 2018-06-15:
Is this a Protocol Designer ticket or a Python protocol ticket or both?

IanLondon, 2018-06-15:
@mcous 

It's a Python API ticket - I propose `robot.pause` command should take a `message` param (right now it's no params) which gets put into the `payload = {'text': message or 'Pausing robot operation'}` of `make_command`... or something like that. And in the JSON protocol executor, it will call robot.pause with the appropriate message. 

By changing `robot.pause` to take an optional message instead of doing something only in the JSON protocol executor, Python protocols can also take advantage of this "show me a message during the pause in the Run commands list" behavior.

PR #2306 (2018-09-18): feat(api): support optional pause message
## overview

Closes #1694 - `robot.pause` supports optional message param. Also fixes unticketed issue where robot.comment causes commands list in Run App to indent.

JSON example (nothing is indented in JSON b/c only low-level methods used):
[image]

.py example (.py uses indentation b/c it uses distribute, etc high-level methods):
[image]

## changelog

* support optional robot.pause message
* fix indentation issue with robot.comment in Run App command log

## review requests

Not sure who to include in a PR like this so I just tagged a handful of pp [... cut]
commit e8056aeb (2018-09-18): feat(api): support optional pause message (#2306)

* fix indentation issue with robot.comment

Closes #1694
closed 2018-09-18 by IanLondon