Opentrons/opentrons #1706: Less dripping tips

howisthisnamenottakenyet, 2018-06-15:
As a user I would like to not have used tips on pipette 1 be dripping across the deck as pipette 2 does work. 

## Acceptance criteria
- Tips are dropped at the end of steps, unless the following step with that same pipette is a "change tips never"

howisthisnamenottakenyet, 2018-06-18:
@IanLondon Weird logic but in the case that Step 2 is "change tips never", should we add (or recommend in an alert) a blow out or touch tip to the last dispense in Step 1? I could still see someone not anticipating that their saved tips might be dripping.

IanLondon, 2018-06-18:
Makes sense to warn them somehow! Is there some way we can make it easy for users to visualize that tips are being carried over?

howisthisnamenottakenyet, 2018-06-18:
My instinct is: we can start with the disappearing tips in tip racks and see if that makes things clear, then add complexity if it doesn't.

IanLondon, 2018-07-12:
Related: #969 (re-opened as a bug, feature regressed)

IanLondon, 2018-07-25:
Also tip dropping on the **last** step for a pipette should happen eagerly - you can think of it like END will count as a "doesn't need tips" step

commit 8ff7af4b (2018-07-26): feat(protocol-designer): make pipettes eagerly drop tips

- drop tips at the end of each step unless the next step for a pipette uses change tip: "never"

Closes #1706
PR #1946 (2018-07-27): feat(protocol-designer): make pipettes eagerly drop tips
## overview

Closes #1706

To make StepCreators composable, and because they individually can't be aware of future steps, they always hang on to their tip at the end. However, this can cause liquids to drip while tips are needlessly carried over. Plus it looks really freaking weird. In practice, the only time tips shouldn't be dropped at the end of a step is if the next step for that pipette has "change tip: never" set -- this means "reuse tips from previous step with this pipette".

So this PR makes the `robotStateTimeline` selector loo [... cut]
commit 9fb0725e (2018-07-27): feat(protocol-designer): make pipettes eagerly drop tips (#1946)

- drop tips at the end of each step unless the next step for a pipette uses change tip: "never"

Closes #1706
commit ef47da5e (2018-08-06): Protocol designer beta staging -> production (#2015)

* feat(protocol-designer): make pipettes eagerly drop tips (#1946)

- drop tips at the end of each step unless the next step for a pipette uses change tip: "never"

Closes #1706

* fix(docker): Switch out dumb-init, add modules tools & udev config (#1952)

- Switch to Resin's default init system
- add avrdude apk & avrdude config file
- add udev rules for modules
Closes #1822

* feat(comp): Add IntervalWrapper to interaction enhancers (#1942)

* fix(protocol-designer): fix styling of pause and mix step items (#1948)

Closes #1947

* feat(pr [... cut]
closed 2018-07-27 by IanLondon