Opentrons/opentrons #1712: Clearer error on "change tips: never" with no tip on pipette

IanLondon, 2018-06-18:
## overview

"Change tips: never" usually means "reuse the tips carried over from the previous step for this pipette". But when there is no previous step for that pipette, it isn't a viable option.

## current behavior

On the first step a particular pipette is being used, "change tip: never" gives an error like "tried to aspirate with no tip on pipette"

## expected behavior

Since the only way to fix the problem is for user to change the "change tip" dropdown on that particular form, maybe this should be a form-level error that comes up immediately when "change-tip: never" is selected, on a step form using a given pipette for the first time in the protocol?

@howisthisnamenottakenyet

howisthisnamenottakenyet, 2018-06-18:
@IanLondon Would it be possible to disable the 'never' option in that scenario?

IanLondon, 2018-06-18:
@howisthisnamenottakenyet 
We couuuld but then if a user deletes upstream steps (Step 1: always, Step 2: never, delete number 1) we still have to deal with it. And if later we allow copying and re-arranging steps, disabling options in general based on where the step is in the timeline makes that process more complicated.

howisthisnamenottakenyet, 2018-06-18:
Fair! I'll add it as an on-blur warning in our errors doc. 

Ideally down the line when we have time it'd be nice to add more complex disabling logic.

closed 2018-07-30 by IanLondon