Opentrons/opentrons #1713: Run App: Unsuccessful connection

pantslakz, 2018-06-18:
As a user, I'd like to know if my robot is unable to connect to app.

## Design
[image]

mcous, 2018-06-19:
@pantslakz I assume this ticket also includes the removal of the existing unsuccessful connection modal?

Kadee80, 2018-06-22:
ping @umbhau @pantslakz, see @mcous question above

pantslakz, 2018-06-22:
Hey, 

This is a mistake on my end - I forgot we already had a modal here

We are currently in process of figuring out when to use modal vs alert banner - would suggest holding off on this ticket until we have rules defined. Modal is working fine for now - apologies if this caused additional work.

mcous, 2018-06-25:
@pantslakz thanks for the update! Moving this ticket back to the backlog for the time being

mcous, 2018-10-18:
Closing because I presume this is covered by existing modal + disabled robots in list via #2434 + new warning banners for unconnectable robots in #2455

closed 2018-10-18 by mcous