Opentrons/opentrons #1715: API's GPIO module swallows write errors

mcous, 2018-06-18:
## overview

If `gpio.py` fails to write a value to a pin, the error is swallowed.

## current behavior

`gpio.py` uses `os.system` to make write and read calls to the filesystem to interact with GPIO. `os.system` does not throw, instead it returns the exit code of the call. `gpio.py` is currently tossing the exit code, non-zero or otherwise.

## steps to reproduce

This is easiest to reproduce with the light toggle hitting virtual smoothie (see #1710)

## expected behavior

A failed write should throw or otherwise communicate the error

mcous, 2019-08-30:
This bug is still present in 3.11.x

sfoster1, 2020-05-12:
It doesn't do this anymore after the switch to `libgpiod` in #5381

closed 2020-05-12 by sfoster1