Opentrons/opentrons #1725: transfer to trash[0] smashes pipette into waste container

5000commas, 2018-06-19:
## 

## current behavior

## steps to reproduce

## expected behavior

mcous, 2018-06-20:
Hi, thanks for the report! Can you please provide us with the following information so we can investigate this issue?

- Robot type
- App version
- Robot software version (if OT2)
- Operating system and version
- Full protocol to reproduce issue

mcous, 2018-06-25:
@5000commas were you able to gather the information requested above?

mcous, 2018-07-12:
@btmorr reproduction protocol:

```py
# multi-channel trash collision repro protocol
from opentrons import containers, instruments

tiprack = containers.load('tiprack-200ul', '9')
plate = containers.load('96-flat', '1')

p300m = instruments.P300_Multi(tip_racks=[tiprack], mount="left")

p300m.pick_up_tip()
p300m.aspirate(25, plate.cols('11'))

# THIS CRASHES THE ROBOT
p300m.dispense(p300m.trash_container)

```

btmorr, 2018-08-22:
The cause for this is that the `dispense` method defaults to the bottom of the well, if a vector is not provided (as opposed to `move_to` commands, for instance, which default to the top of the well). To explicitly dispense into the trash, please use:
```
p300m.dispense(p300m.trash_container.top())
```

willcanine, 2018-08-28:
Talked to @5000commas on Intercom today. Tried @btmorr's fix but didnt solve issue. Code to reproduce:

```
m300.transfer(
transfer_volume,
plate.columns('7'),
trash[0],
new_tip='never')
```

Video of the crash is attached as zip file: [VID_20180828_111526.3gp.zip](https://github.com/Opentrons/opentrons/files/2329469/VID_20180828_111526.3gp.zip)

Conversation link on Intercom: https://app.intercom.io/a/apps/ukpodv2l/inbox/inbox/all/conversations/18203485047

mcous, 2018-10-18:
@willcanine I believe that, according to @btmorr's fix, that transfer should be written as:

```py
m300.transfer(
    transfer_volume,
    plate.columns('7'),
    # this might need to be trash[0].top()
    m300.trash_container.top(),
    new_tip='never')
```

From your reply (can't see the Intercom conversation) it looks like the fix might not have been attempted? @5000commas are you able to confirm?

mcous, 2019-07-09:
Closing because (according to the Intercom conversation) customer issue seems resolved and the transfer code posted above should work. Will re-open if this is not the case

closed 2019-07-09 by mcous

Jonpot, 2024-09-18:
Getting the same issue.

```python
waste = pipette.trash_container.top()
pipette.transfer(volume, plate[well], waste, new_tip='never')
```

Yields TypeError <opentrons.protocol_api.disposal_locations.TrashBin object at 0x0000022D5A00EFB0>