Opentrons/opentrons #1729: Usability: Deactivate Protocol Upload Button During Run

umbhau, 2018-06-21:
As a user, I should not be able to upload a protocol while I have an active run, because it would interfere with the run.

## Acceptance Criteria
- [ ] Deactivate protocol upload button during protocol run

umbhau, 2018-06-21:
fyi @btmorr

Kadee80, 2018-06-25:
In the current app, when a protocol run is in progress:
- the [protocol] button, along with the [calibrate] button in main nav is already disabled. 
- Confirmed this on Sunset. Ran a protocol and had Nusrat connect to Sunset. Upload was disabled on her app as well.

[image]
[image]

closed 2018-06-25 by Kadee80