Opentrons/opentrons #1734: Modules: New 'Modules and Pipettes' Section

umbhau, 2018-06-21:
As a Run App user, I would like to have a dedicated page for my modules and pipettes, so that they're not all crowded on the same page as my robot settings.

## Acceptance Criteria
- [ ] Robot card in robot sidebar has sub-nav for 'Robot' and 'Modules and Pipettes'
- [ ] New 'Modules and Pipettes' page 
- [ ] 'Pipettes' card moved from existing 'Robot' page to new 'Modules and Pipettes' page

pantslakz, 2018-06-23:
[image]

@mcous @howisthisnamenottakenyet @umbhau 
Heyas, some notes re this design:
- side nav panel menu hierarchy is still being worked through - but one option could be to remove the toggle entirely - since it's unclear what it toggles currently, and instead rely on the connect button in robot settings page. 
- removed connection success banner for now. if the connection is successful - it populates in the card. Let's start with that and add more visual cues once we understand the states of modules better.
-

Kadee80, 2018-06-26:
Moving getting and displaying module info into its own ticket. This ticket will be implemented with this screen to avoid being blocked by API endpoint work.

[image]

commit 53a6c13b (2018-06-29): feat(app): Add modules and pipettes settings page

closes #1734
PR #1785 (2018-06-30): feat(app): Add modules and pipettes settings page
## overview

This PR closes #1734

[image]

## changelog

- feat(app): Add modules and pipettes settings page
- add optional exact prop to ListItem 
- add atomic layout components `CardContainer` `CardRow` and `CardColumn`

## review requests

`make -C app make dev OT_APP_MODULES=1` to view AttachedModulesCard 

AttachedModulesCard is not a refresh card yet. Will update the Refresh card once API endpoints are in place. I did a little guessing at what a Modules type might need. Feel free the chime in.

Optional `exact` prop was needed for NavLin [... cut]
closed 2018-06-30 by Kadee80