Opentrons/opentrons #1736: Modules: Update Module Firmware

umbhau, 2018-06-21:
As a Run App user, I would like to be able to update my module's firmware.

## Acceptance Criteria
- [ ] 'Update' button available on 'Modules' card
- [ ] 'Update' button is active if new version is available
- [ ] 'Update' button launches a dialogue box displaying available version number and prompting user to udpate
- [ ] 'Update' button initiates module firmware update
- [ ] User notified of success or failure of update

## Implementation Notes
- Note we will likely build this before update endpoint is available, so will need sane UI for 404.

## Design
- @pantslakz 
- Success and Failure notification UI TBD

pantslakz, 2018-06-23:
@mcous could we treat this the same as we treat robot updates? Let me know if you need screens for this, or if we can just use same modal treatment we have for robot update.

sfoster1, 2020-04-27:
This happened

closed 2020-04-27 by sfoster1