Opentrons/opentrons #1739: Modules: View in Deck Map

umbhau, 2018-06-21:
As a user, I would like to be able to see my modules during deck setup and calibration, so I know where to place them.

## Acceptance Criteria
- [ ] Modules displayed below labware on deck map during deck setup and calibration
- [ ] Slot labels include module type below labware information
- [ ] Visual indication of successful connection (bad connection covered in #1737)

## Implementation Notes
- Dependent on Modules Field in RPC #1654

## Design
- @pantslakz

pantslakz, 2018-06-23:
[image]

umbhau, 2018-07-03:
blocked by #1654

PR #1913 (2018-07-23): feat(app): Show module name over labware on deckmaps
## overview

This is the last remaining PR to close out the current sprint's modules tickets. It adds a module name overlay to any labware that are sitting on modules on the deck.

Closes #1739

[image]

[image]

## changelog

- feat(app): Show module name over labware on deckmaps

## review requests

Follow instructions in #1910 to stub out protocol (session) and actual modules and enable module with `OT_APP_MODULES=1`. Then:

- [ ] Module name overlay appears in labware review modal deckmap
- [ ] Module name overlay appears in labware cali [... cut]
commit c40905b2 (2018-07-23): feat(app): Show module name over labware on deckmaps (#1913)

Closes #1739
closed 2018-07-23 by mcous