Opentrons/opentrons #1740: TempDeck: Real Time Status During Run

umbhau, 2018-06-21:
As a Run App user, I would like to be able to see my TempDeck's status during a run.

## Acceptance Criteria
- [ ] Display current temperature in Run sidebar
- [ ] Display target temperature in Run sidebar
- [ ] Display status (string) in Run sidebar

## Implementation Notes
- Dependent on 'Get Modules / 'Info' endpoint, #1653, including current and target temps
- Will just poll this endpoint

pantslakz, 2018-06-23:
@umbhau are we tackling this in this sprint? I was under impression we'd include temperature status info in the logs, and come back to this idea later.

umbhau, 2018-06-23:
No, almost certainly not. I dropped all the module stories in the sprint, but we will not get that far / Ben will move stuff out after estimation.

pantslakz, 2018-07-24:
[image]

pantslakz, 2018-07-24:
@umbhau @mcous @btmorr just realized we could have more than one temperature module on the deck. Will whip up design to report status on more than one.

umbhau, 2018-07-25:
If I haven’t forgotten something (totally possible), we’re only supporting one temp deck at launch (no way to distinguish between two USB ports/modules).

PR #1932 (2018-07-25): feat(app): Add realtime status TempDeck card to run panel
## overview

This PR closes #1740 - Adds the realtime status card to the run panel (UI only, endpoint still pending). It also adds some mock module data fields for TempDeck and MagDeck see `app/src/http-api-client/modules.js`.

[image]

## changelog

- feat(app): Add realtime status TempDeck card to run panel

## review requests

*For the full module experience:*

comment out line 77, uncomment lines 79-110 in `app/src/http-api-client/modules.js`

uncomment lines 345-351 in `app/src/robot/api-client/client.js`

run `OT_APP_MODULES=1 mak [... cut]
commit 75c8df43 (2018-07-25): feat(app): Add realtime status TempDeck card to run panel (#1932)

closes #1740
closed 2018-07-25 by Kadee80