Opentrons/opentrons #1783: Update labeled value component in PD

howisthisnamenottakenyet, 2018-06-29:
## Acceptance criteria
- Labels for this component are now bold and Aa case 
- Update the labeled value component that displays pipette information on the file details page

## Design
[image]

umbhau, 2018-07-02:
@howisthisnamenottakenyet Should form labels have the same style?

howisthisnamenottakenyet, 2018-07-02:
@umbhau Decided on no. In PD there's a bunch of form fields all packed together and I think that'd make it look more cluttered. 
[image]

umbhau, 2018-07-02:
@howisthisnamenottakenyet sounds good. 

To summarize:
- Value labels will be in **Title Case**, e.g. "Left Pipette"
- Field labels will be in **Sentence case**, e.g. "Tip rack"
- Field 'category headers' (?) will be in **SMALL CAPS**, e.g. "TECHNIQUE", "CHANGE TIP", and "FLOW RATE"

Is that right?

howisthisnamenottakenyet, 2018-07-02:
@umbhau yup! I updated the component pages for form fields and labeled values in zeplin to reflect that

IanLondon, 2018-07-24:
How does this affect Run App? Not sure if I'm following

howisthisnamenottakenyet, 2018-07-25:
@IanLondon The component for pipettes in the run app calibration flow is the same as ours for file details

IanLondon, 2018-07-25:
@howisthisnamenottakenyet Double-checking: this is just changing the `LabeledValue` component instances inside the pipette group component (called `InstrumentGroup`), and NOT changing all `LabeledValue`s across PD + Run App, right?

closed 2019-03-25 by howisthisnamenottakenyet