Opentrons/opentrons #1784: Don't show Export button in PD when there is no protocol to export

howisthisnamenottakenyet, 2018-06-29:
A couple testers were confused as to why the export primary button existed before they made or imported a protocol. 

## Acceptance criteria
- Don't show the export button unless there is a protocol open

## Design
[image]

pantslakz, 2018-07-02:
@howisthisnamenottakenyet curious as to the order of these - 
think it would make most sense to have "Create New" at the top, following by "import" and "export" once option is available. thoughts?

pantslakz, 2018-07-05:
@IanLondon @howisthisnamenottakenyet let's address ordering and text of the buttons in this ticket as well:

Create new
Import protocol
Export protocol

PR #1926 (2018-07-26): feat(protocol-designer): re-order and restyle file sidebar buttons
## overview

Closes #1784

Just some text change and button re-ordering. Also cleaned up old css

## changelog

-restyle file sidebar buttons

## review requests

- buttons look like #1784
commit 4ae1f5b6 (2018-07-26): feat(protocol-designer): re-order and restyle file sidebar buttons (#1926)

Closes #1784
closed 2018-07-26 by IanLondon