Opentrons/opentrons #1794: Website: TempDeck Whitepaper

umbhau, 2018-07-02:
As a potential TempDeck customer on the website, I would like to be able to download the latest TempDeck whitepaper, so that I can get detailed specs and performance characteristics before making a purchasing decisions.

## Acceptance Criteria
- [ ] User able to download TempDeck whitepaper from 'modules' page on Opentrons website.

## Design
- TBD

## Resources
Whitepaper:
https://docs.google.com/document/d/1Mlv2LxL8rmZ9vaOwx3AIiWIqNDltvo2Daxp6-D8AJsI/edit#

mcous, 2018-07-12:
https://github.com/Opentrons/opentrons/commit/1bf8bd7 referenced the wrong issue. Not sure why. Re-opening

mcous, 2019-06-25:
Closed by Opentrons/opentrons-dot-com#99

commit 1bf8bd7e (2018-07-02): feat(update-server): Add restart endpoint to Update Server and shorten restart sleep to 1s (#1793)

Fixes #1794
closed 2019-06-25 by mcous