Opentrons/opentrons #1812: Distribute bug- some destination wells display as mixed

howisthisnamenottakenyet, 2018-07-03:
In a user test Piul was taking Ingredient 1 and distributing it into wells that had two states: empty or Ingredient 2. 

In the case where Ingredient 1 went into empty wells the side panel showed that Ingredient 1 was going into a well with Ingredient 1 in it. 

In the case where Ingredient 1 was going into wells with Ingredient 2, the side panel showed that Ingredient 1 was going into a well that already contained Ingredients 1 & 2. 

It seems to me like distribute was displaying the destination well state after the distribute, not their state before the distribute.

howisthisnamenottakenyet, 2018-07-16:
@IanLondon Feel free to edit this bug ticket to make more sense as to what went wrong technically

IanLondon, 2018-07-16:
@howisthisnamenottakenyet Looks good to me! I put it in the upcoming sprint

PR #1896 (2018-07-20): PD: fix destination well pills in substeps
## overview

Closes #1812 and refactors `generateSubsteps` fn to lessen that fn's responsibilities. This bug was harder to see b/c `generateSubsteps` was managing timeline indicies, whereas now it's just fed the relevant timeline frames.

## changelog

    -calls to swatchColors use `Number` not `parseInt`, warns on non-integer input (NaNs)
    -clean up swatchColors import (no re-routing)
    
    -clarify names of labware/ingred types
    -rename 'substepId' to 'substepIndex' in PD substeps
    -less argument boilerplate in allSubsteps selector
     [... cut]
commit 60481b58 (2018-07-20): fix(protocol-designer): fix destination well pills in substeps (#1896)

-calls to swatchColors use `Number` not `parseInt`, warns on non-integer input (NaNs)
-clean up swatchColors import (no re-routing)
-clarify names of labware/ingred types
-rename 'substepId' to 'substepIndex' in PD substeps
-less argument boilerplate in allSubsteps selector
-remove namedIngredsByLabware selector, convert WellContents to NamedIngreds in getIngredsFactory fn instead
-make generateSubsteps fn work on a single step, no longer is responsible
for mapping across step list itself

Closes #1812
closed 2018-07-20 by IanLondon