Opentrons/opentrons #1836: Timeline Range Error on save bogus volume

b-cooper, 2018-07-09:
## overview

In the Step Edit Form, if I input any integer into the volume field such that the `chunksPerSubTransfer` exceeds the limit of elements in a JS array (4,294,967,295), I receive a Range Error that crashes the app on step save.

## current behavior

## steps to reproduce

Attempt to save a new step with a bogus volume (e.g. 4444444444444444444 µl)

## expected behavior

Save with appropriate form and timeline level warnings and errors present, also don't crash.

IanLondon, 2018-07-09:
Should we set some generous upper limit, and give an error like: "This step would require over 10,000 aspirates and dispenses. Something is wrong. Did you enter your volume for this step correctly? Did you select the right pipette?"

IanLondon, 2018-07-09:
what's the easyfix solution?

b-cooper, 2018-07-09:
Easyfix solution is just to add a protecting conditional in the timeline generator so that it doesn't crash. It can default to Array(upper_limit) if the number exceeds the limit. In terms of warnings, they will almost certainly get a form level warning for exceeding the well capacity, as well as a Timeline level warning from liquid limitations.

closed 2022-09-30 by bndo