Opentrons/opentrons #1849: Breaking Bug: Calibrating tube-rack-15_50ml smashes pipette into rack

rvinzent, 2018-07-10:
## overview
Calibrating the 15-50ml conical rack smashes the pipette into the rack. The dimensions of the container definition appear to be off as the initial guess is about a centimeter too far "south".  This is a tube rack purchased when we first ordered the OT-1 around a year ago.

## current behavior

Calibrating the 15-50ml conical rack smashes the pipette before you have a chance to tell it where the correct position is. Clicking "move to next calibration" will move the pipette to the incorrect position and attempt to lower the pipette all the way to the deck, smashing it into the rack. It was as if it was trying to move to the very bottom of the 15ml conical but guessed the wrong spot.

## steps to reproduce
Attempt to calibrate the `tube-rack-15_50ml` container. I'm not going to try this again, but mine was in position 5.

Version: 3.2.0-beta.3
Firmware: edge-6168d32

## expected behavior

I expect the robot to not crash, first of all. But given that we are calibrating the top of the containers anyway, I see no reason for the robot to even attempt to go lower than the expected height of the container. This way, even if the robot is way off in its initial guess it is still unlikely to crash. I'm afraid to calibrate the `tube-rack-2ml` because it might do the same thing.

Maybe test that calibrating all the documented containers from scratch doesn't crash the robot because at this point I'm a little hesitant to proceed with any more calibrations. 
****

btmorr, 2018-07-10:
Upon first inspection, it appears that the labware definition for `tube-rack-15_50ml` is incorrect, and the other tube racks seem to be correct. One cause for this is that the `tube-rack-15_50ml` is the only one in the labware definition library that is not rotationally symmetric, so the OT1->OT2 trasformation (which deals with the portrait vs landscape different between the two robots) does not transform that labware correctly. 

One practical effect of the portrait->landscape transformation is that OT1 and OT2 seem to disagree on which well is "A1". This difference is invisible on a rotationally symmetric labware, as the relative movements are identical. On this labware, however, it has changed the A1 well from one of the small tubes to one of the large ones, which have different center points. The transformation also shifts the wells incorrectly so that the relative distances between wells is incorrect, even if the user were to begin treating the large tube as A1. I am testing the other tube racks, but this does lead me to believe that any rotationally symmetrical labware are unaffected by this bug.

The height issue seems to be a separate problem. On my development robot, I have deleted my database and started from scratch, and checked the height value for the top of well A1, which is 116.0. I am able to reproduce the incorrect movement (moving the tip to the deck, instead of the top of the well). I'm not sure yet why this is happening, but since I can reproduce I'll be able to debug it.

btmorr, 2018-07-11:
I've reproduced the z-height portion of this bug using this protocol (running in the terminal on the robot):
```
from opentrons import robot, instruments, labware
from opentrons.trackers.pose_tracker import absolute

robot.connect()
robot.home()

tiprack = labware.load('tiprack-200ul', '1')
tuberack = labware.load('tube-rack-15_50ml', '5')

p = instruments.P300_Single(mount='right', tip_racks=[tiprack])

p.pick_up_tip()
wellA1 = tuberack.wells('A1')
rel_top = wellA1.top()[1]
corner = absolute(robot.poses, wellA1)
abs_top = list(map(sum, zip(corner, rel_top)))
p.move_to(wellA1)
```

Notes:
- `abs_top` evaluates to `[149.0, 144.59999999999999, 116.0]`, so p.move_to(wellA1) should go to that position, but instead it is going to the deck (specifically, it's going to the lower-left corner of the well, which is what is tracked in the labware definition).
- `p.move_to(tr)` moves to the lower-left corner of the slot, but offset by the labware's _coordinates value, so it seems like it's going to the corner as tracked in the labware definition, and not correctly adding in the .top()[1] offset (consistent with behavior of the well)
- this works as expected for a '96-flat' plate, and I've deleted my database, so it's not bad calibration data saved from a previous session.
- Checked feature flags, calibrate to bottom and new labware definitions are False.

btmorr, 2018-07-27:
The Z-height portion of this bug was caused by the "z" field of the origin offset portion of the labware definition. Other labware have the "height" of well A1 copied over "z" if it does not exist, but this labware did not get that copy. I generally regard the copy of height into z as an incorrect behavior, but the whole geometry system is getting an overhaul soon so the short-term solution is to utilize this behavior by removing the "z" field from the origin offset field. 

The x-y position errors come from a bug in the data migration script. That script currently works for labware that are rotationally symmetrical. That script should be revised to use slot dimensions to transform each well x-y individually, rather than referencing other wells.

mcous, 2018-09-23:
Ticket was closed by #1959 and fix was released to stable with `v3.4.0`

closed 2018-09-23 by mcous