Opentrons/opentrons #1851: Bug: Home button always disabled

rvinzent, 2018-07-10:
## overview
The `Home` button is always disabled.

## current behavior
I cannot home robot because the button is always disabled.

## steps to reproduce
Version: 3.2.0-beta.3 for Linux
Firmware: edge-6168d32

I first noticed after quitting an unsuccessful calibration (#1849), but I can't be sure that that was the cause. This is also not reproducible on the Windows version of the app. Rebooting the robot and the app has no effect, it is disabled from the start.

## expected behavior
`Home` button should almost never be disabled. If it is disabled for a legitimate reason, an explanation of the reason would be a nice feature.

mcous, 2018-07-10:
@rvinzent to prevent problems, the "Home" button is disabled if either:

a) You are not connected to the robot (i.e. you have not yet clicked "Connect")
b) You are connected but the robot is currently running a protocol

I cannot reproduce the Home button being improperly disabled on `v3.2.0-beta.3`. Can you please confirm that the Home button in your app works if you click "Connect" first?

We just started work on (finally) integrating instructional tooltips into the app, so if you can confirm there isn't a bug, I'll rework this ticket into a feature request for a tooltip on this button

rvinzent, 2018-07-10:
I clicked `Disconnect` and `Connect` again, and the `Home` button is still disabled. There is no protocol attached. I can confirm it's connected because I can turn the lights on and off. I'm not sure if it's relevant that our connection is over USB, but maybe that can help reproduce the problem.

It's almost as if I'm in a perpetual state of running a protocol, which might have something to do with the way I abruptly cancelled the calibration after the crash, but unfortunately I don't remember exactly what I did to cancel.

I was going to try to reupload the protocol and quit again to see if that triggers the "not running a protocol" state, and that made me realize that there is no way to detach you protocol or to quit everything related to that protocol on the calibration or run page. Even restarting the app doesn't remove my protocol. Rebooting the robot, however, does remove the protocol and disables the calibration pages, but it does not enable the `Home` button.

Here is a screenshot of my app view (the protocol page shows nothing attached):

[image]

mcous, 2018-07-10:
Thanks for the screenshot, that's helpful. I will investigate further and continue to try to reproduce

mcous, 2018-07-11:
Hey @rvinzent, we're still unable to reproduce over here. Tried every combination of OS (Mac, Linux, Win 10, Win 7) and connectivity (WiFi, wired) we have. Could you do me a favor and click the "More" button in the bottom left and send me a screenshot of the "App" page?

I can see that your robot is up to date (`3.2.0-beta.3`, which we re-released yesterday as `v3.2.0` stable), but the only thing I can think of is that your app might not be up to date. I definitely think there's a mismatch, because if you app and robot are on the same version, that "Update" button on the "Information" card should read "Update**d**"

rvinzent, 2018-07-11:
That seems to be the most likely cause. The app version appears to still be on `v3.2.0-beta.1`. Unfortunately, due to #1574 I cannot update the app. The latest package in the Ubuntu (16.04) archive is still `v3.2.0-beta.1`  as well.

mcous, 2018-07-11:
@rvinzent sorry about the confusion there! Yes, unlike the OT-1, the app and the API are separate entities with the OT-2. For simplicity we version them together, and generally you're going to want to make sure their versions are lined up. Latest download link is now live on the on our download page as of today: <http://opentrons.com/ot-app>

I'm going to prioritize #1574 because we don't post beta versions anywhere public. I'll also look into starting to post them to GitHub releases, as well.

Closing because home button always disabled is expected in `v3.2.0-beta.1` (actual home functionality was added in `beta.2`)

closed 2018-07-11 by mcous