Opentrons/opentrons #1869: Labware Definitions: Include version integer to track changes

mcous, 2018-07-12:
## overview

As a user, if the base definition of a piece of labware changes, I don't want my previous calibration data to be made incorrect.

## proposal

As we make the switch to splitting base labware definitions from run-to-run calibration data, we're going to run into the problem of the deltas in the calibration data being made inaccurate by changes to the base definition.

In general, once a base definition is "baked", it should never be changed (e.g. in most situations involving changes to labware, creating a new definition with a different name would be the correct course of action). However, we are almost certainly going to run into issues where a given definition is just straight-up wrong (see #1849), and the proper thing to do is to edit the labware definition.

If a labware definition was incorrect in the first place, that means any calibration data is also "incorrect" (or at best, a workaround), so if the base definition changes, the calibration data should be wiped out so as to not introduce incorrect movements.

To accomplish this, we can add a simple `version` integer to the base definitions, and whenever calibration data for a given labware is saved, it should be saved with the same `version` from the base definition. Then, if a labware is ever loaded and the `version` fields between the base and the calibration data do not line up, the calibration data should be discarded.

IanLondon, 2018-07-13:
Seems like a good solution!

Are all these assertions true? Trying to make sure I understand it correctly:

1. We would make `version` a required integer field in labware def JSON schema
2. Whenever a labware definition is updated in a way that could affects saved calibration data, the `version` should be incremented. If something else is updated (eg metadata) it doesn't really matter if you update the `version` or not.
3. Multiple versions of the same labware definition will never co-exist in the repo
4. PD & Run App will use the latest labware definition (which is the only one available to them anyway b/c of 3) for rendering the labware wells.
5. Protocols (both JSON and .py) have no concept of labware `version`. Neither do Run App or PD or Protocol Library. The `version` field is only for calibration cache busting.

So maybe I would want to disambiguate it as `calibrationVersion` or something, as not to confuse it with anything outside of calibration resetting, but I don't feel strongly about that -- we can disambiguate in documentation too.

mcous, 2019-06-25:
Closed by #3469

closed 2019-06-25 by mcous