Opentrons/opentrons #1885: Debug and Admin: Reset Robot

umbhau, 2018-07-17:
PLACEHOLDER As a robot user, I would like to be able to restore my robot to factory settings, so that I erase any bad configuration data without wifi or hands-on support.

## Acceptance Criteria
- 'Restore Factory Settings' button in Robot Settings page, under 'Advanced Settings'
- Clicking 'Restore Factory Settings' button erases or sets to factory defaults (?) the following:
-- Deck calibration
-- User labware calibrateion
-- TBD
- Explanatory copy (and warning dialogue box): TBD

## Background
- This is a request from the support team (and customers); there are currently quite a few ways to end up with bad (e.g.) deck calibration data, and clearing it can be quite hard if customer is not on wifi.
- Note there is some risk in clearing mount-offset, since user cannot generally recreate it themselves

umbhau, 2018-07-17:
@andySigler @btmorr 

I've gotten a few requests for a 'reset' feature from the support team, particularly to clear out bad deck calibration or mount offset data.

Could use your help thinking through what specific configs we'd want to reset (or restore to defaults), what 'reset' or 'restore' would mean for each, and any risks (e.g. not being able to tip probe after erasing a mount offset).

pantslakz, 2018-07-27:
does this require restart?

umbhau, 2018-08-07:
@btmorr @Laura-Danielle to discuss reset specifics (what data to delete). include support.

@pantslakz this will require a restart

mcous, 2018-08-15:
Request/response proposal as requested. No strong opinions on actual endpoint path, just picked something for the purposes of writing this comment. Feel free to use a different path.

```js
// GET /settings/reset/options
// 200 OK
{
  deckCalibration: {
    name: 'Deck Calibration',
    description: "Description of Deck Calibration data and why you'd clear it"
  },
  tipProbe: {
    name: 'Tip Probe',
    description: "Description of Tip Probe data and why you'd clear it"
  },
  labwareCalibration: {
    name: 'Labware Calibration',
    description: "Description of Labware Calibration data and why you'd clear it"
  },
  bootScripts: {
    name: 'Boot Scripts',
    description: "Description of Boot Scripts data and why you'd clear it"
  }
}

// 500 ISE - Generic 500 handler is fine

// POST /settings/reset
// Request
{
  deckCalibration: true,
  tipProbe: false,
  labwareCalibration: false,
  bootScripts: true
}

// 200 OK
{}

// 400 Bad Request (For example)
{
  "message": "topProbe is not a valid reset option"
}

// 500 ISE - Generic 500 handler is fine
```

sfoster1, 2018-08-15:
Further clarification - if an option is not included I'll assume `false`. Let me know if you want another behavior.

umbhau, 2018-08-17:
@btmorr @pantslakz @sfoster1 @Kadee80 

Proposal from @Laura-Danielle to add an option to delete the database (/data/user_storage/opentrons_data/opentrons.db).

Thumbs up from me -- any objections?

sfoster1, 2018-08-17:
@umbhau Do you have suggested wording for the description fields in `GET /settings/reset/options`?

Kadee80, 2018-08-17:
@sfoster1 Heres what I have in the UI currently. Keep in mind it will possibly change next week once Pantea's had eyes on this.

Reset calibration of pipette to deck
Erase custom labware calibration
Erase tip probe data
Erase custom boot scripts

Kadee80, 2018-08-20:
@sfoster1 @btmorr (or whoever picks up this ticket)
I amended @mcous comment above to reflect a data structure closer to what we have for GET /settings

```
// GET /settings/reset/options
// 200 OK
[
  {
    id: 'deckCalibration'
    name: 'Deck Calibration',
    description: "Reset calibration of pipette to deck"
  },
  {
    id: 'tipProbe'
    name: 'Tip Length',
    description: "Erase tip probe data"
  },
  {
    id: 'labwareCalibration'
    name: 'Labware Calibration',
    description: "Erase custom labware calibration"
  },
  {
    id: 'bootScripts'
    name: 'Boot Scripts',
    description: "Erase custom boot scripts"
  }
]
```

PR #2082 (2018-08-20): feat(api): Robot Reset
## overview

 Add endpoints at GET /settings/reset/options and POST /settings/reset to, respectively, query what reset options are available and reset any subset of those options to their factory configurations. This allows us to reset only the parts of the configuration that a user might want, and avoid resetting those configurations that are required for operation and difficult to change.

Closes #1885  

GET /settings/reset/options

```
200 OK
[
    {
        "id": "deckCalibration",
        "name": "Deck Calibration",
        "description": "Reset calibration of pipe [... cut]
closed 2018-08-20 by sfoster1