Opentrons/opentrons #1904: Robot Reliability: Reevaluate Date Types and Locations

umbhau, 2018-07-20:
As a user, I would like to reduce data read/write from the thumb drive, to mitigate impact of thumb drive issues.

## Acceptance Criteria
- Data should be written to both /data (persistent data managed by resin) and thumb drive
- Data should be read from /data
- If data not found in /data, read from thumb drive and copy to /data
- if thumb drive also empty, use defaults.

## Implementation Notes
- We hypothesize that thumb drive may not be able to handle high volume of read/writes, and that this may be contributing to thumb drive going read-only
- Need to define what data is in scope for this

umbhau, 2018-07-20:
@btmorr fyi, as per our discussion

btmorr, 2019-06-25:
This has been obviated by other work

closed 2019-06-25 by btmorr