Opentrons/opentrons #1955: 300 uL tip racks are referred to as 200 uL tip racks

howisthisnamenottakenyet, 2018-07-30:
Currently we are referring to ur 300 uL tip racks as 200 uL tip racks in both run app and PD.

pantslakz, 2018-07-30:
@btmorr

btmorr, 2018-07-30:
We should replace those references in documentation and apps with "opentrons-tiprack-300ul"

IanLondon, 2018-07-31:
I can get around this quickly for now by just renaming the dropdown, but...

@btmorr is `"opentrons-tiprack-300ul"` labware supported in API yet, or is it blocked by labware migration issues?

IanLondon, 2018-07-31:
Actually this would be a pretty ugly hack, it would be 300 in the dropdown and 300 in the add labware modal, but 200 in the InstrumentGroup's "Tip Rack" LabeledValue, and also default to "TIPRACK 200 UL (1)" on the deck. I'm gonna call this blocked until there's API support for `"opentrons-tiprack-300ul"` labware

PR #2223 (2018-09-13): feat(protocol-designer): replace 200µl tiprack with 300µl tiprack
## Overview

Remove the old 200µl tiprack as an option in labware and replace it with the 300µl tiprack.

Closes #1955
commit 8a8fc0f7 (2018-09-13): feat(protocol-designer): replace 200µl tiprack with 300µl tiprack (#2223)

Remove the old 200µl tiprack as an option in labware and replace it with the 300µl tiprack.

Closes #1955
closed 2018-09-13 by b-cooper