Opentrons/opentrons #1968: Pipette Not Recognized

umbhau, 2018-07-31:
## Expected Behavior
- Pipette name should appear in Run App when properly plugged in

## Observed Behavior
- Run App reports that pipette is not recognized

## Steps to Recreate
- TBD

umbhau, 2018-07-31:
@mcous @pantslakz 

FYI -- recreated this bug b/c the issue reemerged on one of @pantslakz customer calls yesterday.

mcous, 2018-08-03:
@umbhau and @pantslakz this ticket needs more information. On what screen does the app "not recognize" the pipette? What version of the app has this issue?

If the answers to those questions are "Calibrate > Pipette Calibration" and `v3.2.0`, then this issue was fixed for most cases by #1830 (released in `v3.3.0-beta.0`) and fully fixed by #1895 (released in `v3.3.0-beta.1`)

pantslakz, 2018-08-03:
Confirming that answer is Calibrate > Pipette Calibration" and v3.2.0,

> On Aug 3, 2018, at 8:29 AM, Mike Cousins <notifications@github.com> wrote:
> 
> Calibrate > Pipette Calibration" and v3.2.0,

umbhau, 2018-08-07:
@mcous @pantslakz Any objections to my closing this one?

As far as I know we had three different issues, all of which are now resolved:
1) Bug in FW version 3.1.2
2) Bug in Run App version 3.2.0 (calibrate -> calibrate pipettes)
3) Pipettes not flashed

umbhau, 2018-08-09:
Closing this issue -- believe all known root causes are fixed.

closed 2018-08-09 by umbhau