Opentrons/opentrons #1985: Liquid in wells tooltips (iteration 1)

howisthisnamenottakenyet, 2018-07-31:
## Design
https://docs.google.com/document/d/1MRViHcFCLb9jrPEoW4YjsfbLXAin7-bfAYTy7etpHys/edit#bookmark=id.mu8o6r2dctal

howisthisnamenottakenyet, 2018-08-07:
@pantslakz @umbhau 
These are the designs I have for the Sprint 36 'View Liquid Name in Wells and Ingredient Sidebar' part of our roadmap doc.

umbhau, 2018-08-07:
Hi @howisthisnamenottakenyet, should well tooltip appear in well selection modal during both deck setup and step edit mode?

Otherwise -- made a minor formatting edit but looks good!

pantslakz, 2018-08-08:
@howisthisnamenottakenyet given we're dealing with small wells and lots of info packed in a small space, would suggest adding a small arrow to the tooltip, so there's no confusion as to what it's referencing. i.e.:
https://www.google.com/url?sa=i&rct=j&q=&esrc=s&source=images&cd=&ved=2ahUKEwiG_Iecht3cAhVvMewKHW0UCTgQjRx6BAgBEAU&url=https%3A%2F%2Feclipsesource.com%2Fde%2Fblogs%2F2013%2F09%2F27%2Fcoming-up-rap-2-2m2-with-sexy-new-tooltips%2F&psig=AOvVaw0DKf4MFUA-arkFd-ykky-7&ust=1533803696061932

howisthisnamenottakenyet, 2018-08-08:
@umbhau I don't see a reason why it shouldn't be there for both. 

@pantslakz Ok! I'll make a separate components update ticket for that.

pantslakz, 2018-09-07:
@howisthisnamenottakenyet do we have a ticket for iteration 2?

umbhau, 2018-10-03:
Closing in favor of #2409

closed 2018-10-03 by umbhau