Opentrons/opentrons #1999: App should generate and save a unique user ID for intercom

mcous, 2018-08-02:
## overview

As an app user, I would like my Intercom conversation history to be preserved even if I re-install the app.

## current behavior

If the user re-installs the app, their intercom conversations and history are lost.

## implementation details

Similarly to how we generate a unique app_id for analytics purposes, we should generate another unique ID for use as the user ID. **This ID should be different than the analytics ID** so we cannot link specific users to specific analytics events.

pantslakz, 2018-08-02:
@mcous I should note that this is happening even if user just closes app and restarts it - they lose their prior convo.

mcous, 2018-08-02:
@pantslakz did not realize that was the case. Updated ticket and will prioritize higher, accordingly

PR #2004 (2018-08-06): feat(app): Add persistent unique user ID to intercom data
## overview

This PR updates the intercom configuration to attach a unique user ID (generated on first open of the app) that is stored on the filesystem to persist between app openings and re-installs. This _should_ allow users to keep their conversations going.

It also lays a little bit of groundwork for users to be able to input their name and email for support purposes.

Closes #1999

## changelog

- feat(app): Add persistent unique user ID to intercom data 

## review requests

I separated support from analytics to make sure we don [... cut]
commit 0a47d646 (2018-08-06): feat(app): Add persistent unique user ID to intercom data (#2004)

Closes #1999
closed 2018-08-06 by mcous