Opentrons/opentrons #2005: Disallow touch tip if you have a trough selected

howisthisnamenottakenyet, 2018-08-03:
## Acceptance criteria
- [ ] If you select trough as your labware in the advanced settings for aspirate/dispense touch tip is disabled
- [ ] If you select touch tip in advanced settings then go and set labware to trough, touch tip is automatically de-selected and disabled
- [ ] Nice to have: when disabled get a tooltip that says "this setting is incompatible labware"

IanLondon, 2018-11-29:
Tested in aef8cc8284459342f4ca408cd9796a58ad680dc1 - instead of causing a collision with touch tip on trough, the API now makes the touch tip on a trough cause no x/y movement. With an explicit touch tip z-offset set in PD, it will move to the specified z but there's no collision.

After the robot singleton refactor, touch tip should work correctly...?

pantslakz, 2018-12-05:
@IanLondon we may want to disallow touch tip option for some labware

closed 2022-09-30 by bndo