Opentrons/opentrons #2021: Exit well selection mode for step + ingredient forms by clicking on modal outside of the well

howisthisnamenottakenyet, 2018-08-07:
As a user I'd like to be able to easily exit the well selection modal

## Notes
Follows a similar pattern for exiting out of move mode while placing labware. If you click on something outside of what is 'selectable' then you exit

## Acceptance criteria
- [ ] Clicking outside of the labware in well selection mode closes the well selection modal

## Design
[image]

umbhau, 2018-10-03:
Further design questions:
- Are there modals you can or can't click out of? 
- If so, which modals fall into which categories, and how can we visually differentiate the two situations to users?

pantslakz, 2018-12-06:
@howisthisnamenottakenyet looks currently implemented for 'steps > well selection', but not for 'starting deck state > labware > name & liquids". 

Also we should add 'esc' as a way to go back as well. please update ticket and check to see if there's any other duplicates.

closed 2022-09-30 by bndo