Opentrons/opentrons #2028: Release Notes

umbhau, 2018-08-08:
As an Opentrons user, I would like to know what features and changes are in a software release before I install it.

## Acceptance Criteria
- Release notes displayed in both App Update and Robot Update modal (the modals that are displayed when you click the 'update' button in App or Robot Settings pages.
- Release notes are pulled directly from GitHub release notes (tag description). 
-- Note we should use the 'description' but ignore the 'title'
-- Note release notes will be the same for Run App, Robot, and FW (they are versioned together, so release notes should describe changes to any of them)
- Modal includes a link to the release notes in GitHub
- Modal should be able to gracefully accommodate body text that is longer than the modal height

## Design
- TBD @pantslakz

umbhau, 2018-08-08:
@mcous please let me know if this aligns with your thinking on release notes implementation before @pantslakz starts on design.

mcous, 2018-08-08:
@umbhau after spending an amount of time reading `electron-builder` documentation, I think going off GitHub releases will be difficult based on our existing repository and tooling setup. We have other avenues available, so we should discuss

umbhau, 2018-08-08:
@pantslakz spoke offline with Mike; we have open questions about implementation (may not use GH), but these should not impact design requirements. 

Removed implementation details from user story; from a design POV, we can plan on having access to release note copy in the app (so we can, e.g., display them in the 'update app' modal).

pantslakz, 2018-08-09:
@mcous @umbhau where can I access a list of updates, to help me use as dummy text in design?

umbhau, 2018-08-26:
@pantslakz please add the designs we reviewed to this user story (aiming to slot it in this sprint)

pantslakz, 2018-08-27:
done! 

@howisthisnamenottakenyet didn't add text about rebooting, because in update flow it lets user know they can cancel but changes will not take effect until rebooted.

mcous, 2018-09-24:
Closing because the app update modal now displays release notes and the robot update modal has a separate ticket (#2353)

closed 2018-09-24 by mcous