Opentrons/opentrons #2029: RFC: Standardized title components in Components Library

Kadee80, 2018-08-08:
## overview

Currently, we have a few repeating title styles throughout the App and PD -  and a few one offs.

All titles share the same size
- font-size: header `--fs-header`

Titles have 2 weight variants
- font-weight: semibold  `--fw-semibold` (modals)
- font-weight: regular `--fw-regular` (cards)

Alert Modal titles always have an icon and grey background

Wizard Modal titles are always uppercase

# Outstanding Questions:

1-  Do `Card` and `Modal` components always have title?

2- Should the `CardTitle` or `ModalTitle` render as an (optional?) prop of `Card` or `Modal`? 
Or should the `CardTitle` or `ModalTitle` be their own `Title` atoms that are passed in as children to cards, modals, or other instances of titles with that style?

# Title assets across projects

PD- modal labware select
[image]

PD- 'section' title
[image]

APP- card title
[image]

APP -  Modal Page (Wizard) title
[image]

COMP LIB - Alert Modal title
[image]

COMP LIB-  Generic modal title
[image]

howisthisnamenottakenyet, 2018-08-08:
1) All modal titles and headers are now in Aa instead of AA. 
2) All modals have titles. All cards in PD have titles too. I can't speak certainly on run app. @pantslakz ?
--
3) I've made titles semibold in modals but regular weight in cards because when I initially designed the robot dashboard having a semibold title in each card felt like too much. That was a total off the cuff choice and I'm down to change it.

Kadee80, 2018-08-08:
@howisthisnamenottakenyet  @pantslakz 
1- If all titles are now Aa, instead of AA am I correct in assuming the titles in ModalPages (Wizards) should be changed to match this?

2- All cards have titles in the app, the only modals that dont have titles are 1-offs which would be updated to be a- AlertModals or b- Plain modals with titles. And example of this is:
[image]

3- I personally prefer the lighter font weight for the cards as well.

Kadee80, 2018-08-08:
@mcous @b-cooper 
Based on Morgan's comments above, 
- all cards have titles
- all modals have titles
- all card title styles should be the same
- all modal titles should be the same

Based on this, how do we feel about keeping `title` as a prop in both cards and modals as opposed to making title components?

howisthisnamenottakenyet, 2018-08-08:
@Kadee80 I vote yes for changing all modals to have titles, and titles that are Aa. 

However I'm sure Pantea's far more aware of the nuances of Run App modals and can better speak to their needs. If there are currently modals without titles then that could be an issue.