Opentrons/opentrons #2030: Debug and Admin: Reset Robot (App)

umbhau, 2018-08-08:
As a Run App user, I would like to be able to reset key robot configs to factory settings, as it's often a helpful troubleshooting step and currently has to be done from CLI.

## Acceptance Criteria
- User able to reset the configs/files specified in #1885 (TBD). 
- User provided with name and brief description (?) of each config
- User able to select which configs/file to reset, and to reset them all at once
- User given a warning that this cannot be undone, and a chance to cancel or confirm. User also informed that robot will reset and that they will have to reconnect to robot.

## Design
- TBD @pantslakz 

## Implementation Notes
- Dependent on #1885 
- Design should be flexible to accommodate additional configs/data in the future.

umbhau, 2018-08-08:
Few questions for @btmorr:

1 ) any reason we can't let the user opt to reset one or many configs at the same time (e.g. click reset and delete tip probe and deck calibration, but not labware calibration)?
2) besides deck calibration, tip probe, and user labware calibration, anything we want to include? (will also get input from support team)?
3) any thoughts on how we want to break down tip probe? e.g., can we allow the user to independently reset tip length, mount offset, etc.?

Kadee80, 2018-08-22:
@umbhau re: conversation with pantea

- Moved the logs and reset buttons to the top of advanced settings card
[image]

- Updated settings Names and Descriptions in the API response
- Changed from toggles to checkboxes 
[image]

- Clicking reset button successfully resets/removes the proper files. But a robot restart is required, much like a robot update
[image]

- Once restart is initiated, user is informed that this may take a few minutes and is prompted to close the modal
[image]

PR #2088 (2018-08-23): feat(app): Add reset robot UI with options
## overview

This PR adds reset robot button to advanced settings and reset robot with options modal(s)

Closes #2030

[image]
[image]
[image]
[image]

## changelog

- feat(app): Add reset robot UI with options

## review requests

Please make sure it works on actual robots.
commit c242de94 (2018-08-23): feat(app): Add reset robot UI with options (#2088)

Closes #2030
closed 2018-08-23 by mcous