Opentrons/opentrons #2031: Protocol delay working like a pause

umbhau, 2018-08-08:
Protocol delay currently being represented as a protocol pause in the run app.

## Expected Behavior
- During a protocol delay (pipette.delay(time)), user should see an alert bar 
- Alert bar should say 'Delaying for X time'
- Run clock should continue incrementing during delay
- User should not be able to resume protocol

## Current Behavior
- Robot displays a 'pause' alert bar
- Run clock stops incrementing
- User able to resume protocol

## Steps to Recreate
- Load and run a protocol with the following command: p300.delay(minutes=3)
- Observe robot pause during delay

Laura-Danielle, 2018-08-29:
Has this been fixed yet? @umbhau @sfoster1

mcous, 2019-12-03:
Upon investigation and some timeboxed experimentation, I'm ice-boxing this feature request until RPC is replaced. Delay working via `pause` and `resume` was an intentional API decision, and unwinding that to implement these features as requested exposes too many technical and UX edge cases.

Rather than risk implementing something buggy (or introducing new bugs in existing behavior), we'll keep this ticket in mind as table-stakes for the new Sessions API

Matt-Zwimpfer, 2021-06-03:
Fixed in Protocol Engine

closed 2021-06-03 by Matt-Zwimpfer