Opentrons/opentrons #2039: Collision May Occur while Calibration

simonchih, 2018-08-10:
## overview
If customized labware actual high more than wells depth. Then collision may occur while calibrate this labware. 

## current behavior
Collision may occur while calibrate customized labware that actual high more than wells depth.

## steps to reproduce
1. Put customized labware which high is 20 mm
2. Try to create new labware while depth small than 20 mm, e.g. depth=10:
`
...
custom_plate = labware.create(
    '3x6_plate',                    # name of you container
    grid=(3, 6),                    # specify amount of (rows, columns)
    spacing=(12, 12),               # distances (mm) between each (row, column)
    diameter=5,                     # diameter (mm) of each well on the plate
    depth=10,                       # depth (mm) of each well on the plate
    volume=200)
...
`
3. Open OT2 App
4. Connect robot and upload step 2's protocol
5. Go to Calibrate in OT2 App
6. Collision may occur while calibrate step 2's customized labware

## expected behavior
Collision won't happen.

simonchih, 2018-08-11:
This defect happen with wells depth of labware is smaller than labware height. It can reproduce on version 3.3.0-beta.1.

mcous, 2018-08-13:
Sorry for the inconvenience! After some digging, this appears to be an inherent limitation of the `labware.create` method. I believe we could work around this by using a different method of constructing the custom labware. @btmorr you mentioned something about a builder-style interface we have available?

jlmaccal, 2018-11-02:
Has there been an action on this issue? We were recently bitten by this as well. Every piece of custom labware that we've needed to add has wells that don't reach all the way to the bottom. When creating labware, it should be possible to specify both the well depth and the height of the top of the plate separately.

Stefan1902, 2019-05-23:
I don't see height as a problem. A bigger issue is, that if you have your self created labware on the left side, and try to calibrate with the right pipette, the OT2 runs into the left limit of the axes. So the first calibration is always wrong, if you don't home again.

closed 2020-05-07 by sfoster1