Opentrons/opentrons #2042: Lag in the middle of a protocol

rvinzent, 2018-08-10:
## overview

Protocols sometimes start to lag on all actions of the robot, even in the middle of grabbing a tip. It gets worse and worse until it takes over 5 minutes to eventually get through a single transfer.

There is also serious lag when uploading a protocol. It usually takes about 20 seconds to upload a script with a few hundred transfers, which is a couple orders of magnitude slower than the OT-One.

## current behavior

There shouldn't be any difference in performance between the beginning and end of the run.

[Here](https://media.giphy.com/media/kiXWpcWcS43Qy9Zr93/giphy.mp4) is when we noticed the start of the behavior.

[Here](https://media.giphy.com/media/9x5coZRzFeVin0iKMG/giphy.mp4) is a few minutes later, completely stopped.

## steps to reproduce
App version: `3.2.0`
Robot version: `3.2.0-beta3` (I still can't update the robot server from the app)

This was occurring during the following transfer where `plate` is a `96-deep-well` and `output` is a `96-PCR-flat`:

    p10.transfer(
        4,
        plate[well],
        output[well].bottom(),
        new_tip='always',
        blow_out=True)

which is nothing out of the ordinary compared to every other transfer we do.

## expected behavior

Should be little to no lag, especially when the robot is connected by USB, so latency is not a factor.

mcous, 2018-08-10:
Hey @rvinzent, you've put a few different things in this issue. Tickets are more likely to be addressed if they're scoped to one specific thing, mostly because we can lose track of them. To narrow this particular ticket down:

> There is also serious lag when uploading a protocol. It usually takes about 20 seconds to upload a script with a few hundred transfers, which is a couple orders of magnitude slower than the OT-One.

We understand that this is really annoying. The background is that with the OT1, Python protocols were simulated on your machine, and during a run, your machine sent GCode over a USB-to-UART cable directly to the robot's motor controller board. The OT2 is entirely different in that your computer is communicating via IP with the OT2's onboard computer, which is now the thing that's "simulating" protocols and talking to the motor controller. The OT2's computer is significantly less powerful than your laptop, which is why it takes so much longer to simulate.

(As an aside: the "USB" on the OT2 is actually a USB-to-ethernet adapter plugged into the onboard computer's ethernet port, so there's effectively no speed difference between "USB" and WiFi).

We are _very_ aware of this issue and working hard to fix it, but unfortunately due to how the system  was initially designed, it's a large refactor / cleanup project.

> App version: 3.2.0
> Robot version: 3.2.0-beta3 (I still can't update the robot server from the app)

This is odd, but I think it might be because `3.2.0` and `3.2.0-beta.3` are [exactly the same code](https://github.com/Opentrons/opentrons/blob/edge/CHANGELOG.md#320-2018-07-10); we just bumped the version number to promote `beta` to `stable`. If this persists through our next release, though, I'd really appreciate it if you could file an issue with the relevant details.

> Protocols sometimes start to lag on all actions of the robot, even in the middle of grabbing a tip. It gets worse and worse until it takes over 5 minutes to eventually get through a single transfer.

This sounds like something we've observed in house, too. Based on your description, I wanna say it sounds like a memory leak of some sort...

- If you haven't already, are you able to send us the protocol that is exhibiting this behavior?
- Does the behavior happen consistently? i.e. every time you run this protocol

Matt-Zwimpfer, 2021-06-03:
Resolved in [#2105](https://github.com/Opentrons/opentrons/issues/2105).

closed 2021-06-03 by Matt-Zwimpfer